Snapmaker/snapmaker_cnc_post_process #19: aspire or vcarve pp not processing the tool paths properly

gysnst2, 2021-04-24:
I doing some vcarve practice files.  and when I set the work origin, the ran my boundary every this was great. I ran the file and it started to cut circles.  not the file. I let is run for a bit until it started to cut threw my snapmaker hold-downs. i stopped it and loaded the gcode into luban and omg I was surprised at what I saw.  so then I reinstalled the postprocessor and recalculated the file and loaded it into luban and now at least it shows it being a good process it is not cutting the entire file.  I have tried to recalculate the file several time with different outcomes.

gysnst2, 2021-04-27:
I swear no-one reads these

MickeEKlund, 2021-04-28:
**So, I got this solved! :-)**   

I'm using VCarve Desktop 10.514. The curves looks like a mess in Luban but cuts correctly, just transfer to SnapMaker and go (do yout boundry check first, just in case).

I took the PostProcessor for Aspire (not vCarve) and changed two lines:

VAR ARC_CENTRE_I_**ABS**_POSITION = [I|A| I|1.3]
VAR ARC_CENTRE_J_**ABS**_POSITION = [J|A| J|1.3]

I replaced INC with ABS and so far it has turned out great! Works like a charm! A small sign that took 4 hours yesterday has run in like 30 minutes!

PP-File: [https://forum.snapmaker.com/uploads/short-url/eaPc8GWblYMozc5sPftLI8Lnk18.zip](url)
Thread on Snapmaker Forum: [https://forum.snapmaker.com/t/cnc-vcarve-desktop-start-and-stop/15908](url)

gysnst2, 2021-04-28:
Boundary checks are always good but then it cut threw my snapmaker hold down.  And in vetric it shows a full shogun being cut but when it loaded in LuBan only part of the project is being carved.  Or it's crazy carving circles

MickeEKlund, 2021-04-28:
My did too, but after replacing the PP file it’s working 100%

BR
Mike

________________________________

Mikael Eklund
Kesuddens Konsult AB
Röksvampsgatan 17
749 42 Enköping
***@***.******@***.***>
0702-600 860

artemician, 2021-04-28:
Sorry I didn't see this soon enough or I could've helped. I use aspire no problem. When I got the post processor file off GitHub through the GitHub desktop, for whatever reason moving the file from the default GitHub folder into the aspire folder corrupts it and causes the pp file to not work right.

As far as vCarve goes, vCarve is a segment taken from aspire so that if people only need the vCarve features and not the entire Aspire collection or want to buy the full version of vCarve without the high price tag of Aspire, then they can. There should be no difference between doing this project through vCarve vs Aspire unless some settings you didn't know about weren't set correctly because, again, Aspire includes everything whereas vCarve is just a segment taken from Aspire.

I bought Aspire and have spent months learning, tweaking, tuning and practicing to make it run perfectly with the Snapmaker 2.0 and once the learning process is done, Aspire works really well for the 2.0.

Aspire can be overwhelming at first because of the sheer amount of features and what it can do. If you're new to CNC it's a good idea to start with the different standalone programs (vCarve, Cut3d, Cut2d etc) so you don't get confused as to what feature is best for what project, then ease into Aspire if you want to do more.

gysnst2, 2021-04-28:
> **So, I got this solved! :-)**
>
>
>
> I'm using VCarve Desktop 10.514. The curves looks like a mess in Luban but cuts correctly, just transfer to SnapMaker and go (do yout boundry check first, just [...]

Do you have a copy of your new set of files.  Also when I click on your links I get 404 error

gysnst2, 2021-04-28:
> Sorry I didn't see this soon enough or I could've helped. I use aspire no problem. When I got the post processor file off GitHub through the GitHub desktop, for whatever reason moving the file from [...]

I agree with you I have vcarve full version I paid a prety penny for it if I remember 1000$ or so.  I'm not new to vcarve. But I'm am struggling with LuBan and the vcarve postprocessor.  LuBan is not the worst but doesn't help us for cnc up until 3.14 LuBan just crashed for me.   I have used fusion with some success but fusion is not the workflow I like vcarve is pretty easy for me.  I use vcarve on my v1

MickeEKlund, 2021-04-29:
Copy (the actual text, **not** right click and copy link) and paste the link in a new window and it will work...  [https://forum.snapmaker.com/uploads/short-url/eaPc8GWblYMozc5sPftLI8Lnk18.zip](url) 
Not sure why github breaks the link, I'm pasting it correctly I think....

MickeEKlund, 2021-04-29:
[Snapmaker350ABS.zip](https://github.com/Snapmaker/snapmaker_cnc_post_process/files/6398291/Snapmaker350ABS.zip)

gysnst2, 2021-04-29:
I used your zip file it made a tiny bit of difference 

Maybe pm me your messenger or something we can compare notes a little more

[image]
This is a 1/2 vbit pretty stock settings

[image] this one is with a 1/2 vbit .25" dept set their is no reason for it not to cave the entire sign

MickeEKlund, 2021-04-30:
My Luban look like a mess too but the SnapMaker cuts perfect anyways...  If I open the file in NCviewer it look OK!

I just load the G-code into Luban and sends it via WiFi anyways. Cuts just like it looks in NCViewer!

[image]
[image]

gysnst2, 2021-05-03:
[image]

[image]

[image]

I tried the profile it's a bit better but doesn't fix my issues.

pistrak, 2024-09-13:
Я установил в Aspire пост процесс и и мой snapmaker a350 заработал.
[image]

pistrak, 2024-09-13:
[https://forum.snapmaker.com/uploads/short-url/eaPc8GWblYMozc5sPftLI8Lnk18.zip](https://github.com/Snapmaker/snapmaker_cnc_post_process/issues/url)