Snapmaker/snapmaker_cnc_post_process #20: Generated spindle speed is not accepted by Snapmaker controller

valexpi, 2021-05-07:
The PP script generates a gcode where the spindle speed is written like M3 S12000 (RPM).
The snapmaker firmware does only accept spindle speed written in percent like P100 (which is 12000 rpm / max speed) or P50 (is 6000 rpm).

My personal workaround was to use this line in the pp file:
VAR SPINDLE_SPEED = [S|A| P|1.0]
And configure the speed in tool in percent (e.g. 100 or 75 or 50...)

mikimite, 2023-01-11:
Hi, just realized the same. In the controller M3 Gcode implementation, either P or S parameter are processed.
[Here the code](https://github.com/Snapmaker/Snapmaker2-Controller/blob/514e706dcd68ddcb6ceea9b6a17338523a1b8dac/snapmaker/src/gcode/M3-M5.cpp#L68)

But the issue is that the S parameter exported by aspire is rpm, and the S parameter expected is pwm duty cycle 0-255

This should be addressed in the firmware itself, but maping rpm to power in the postprocessor is a good work arround

bjornhaa, 2024-11-08:
is nothing happening in this project? Luban is pretty much useless for CNC for anything but the trivial projects, and this bug is kinda making Fusion 360 useless in combination with Snapmaker