Snapmaker/snapmaker_cnc_post_process #26: Fusion360 rotary cnc with origin placed properly at the top of the stock leads to errors in B code.

connwork, 2023-09-02:
All drivers and programs are updated to latest versions. 

Generating gcode through Fusion360 using "Wrap Toolpath" leads to B rotation that is out of line with the rest of the gcode when the origin is placed at the top of the stock. 

Path shown on Fusion360: 
[image]

Path shown on Luban: 
[image]

Gcode: 
[try2.txt](https://github.com/Snapmaker/snapmaker_cnc_post_process/files/12502156/try2.txt)

lavoieje, 2023-09-13:
I think Fusion 360 can only perform post-processing correctly if the axe is in the center of the cylindre.

connwork, 2023-09-13:
> I think Fusion 360 can only perform post-processing correctly if the axe is in the center of the cylindre.

I've been told by the people over at fusion360 that this is a problem with the controller https://forums.autodesk.com/t5/fusion-360-manufacture/modified-fanuc-post-processor-for-4th-axis-cnc-outputting/m-p/12238486#M145588

Here is the quote:

"Its not a software limitation, its a limitation of the machine controller.   If the control doesn't support tilted work plane(G68.2) and/or TCP(G43.4) then you must program from center of rotation, this is extremely common on 4 axis machines as most do not have these features enabled.   I think you will find all other cam software will have the same issues"

Is there any way we can enable these gcode features?

connwork, 2023-09-13:
> Snapmaker is based on Malin GCode and actually, both commands are not defined in it. I'm not sure Snapmaker developers plan to improve Marlin.

This is not a new problem. Many people have already supplied solutions. A quick google search of "M43.4 marlin" showed several results, such as this link: https://derandere.gitlab.io/Marlin2ForPipetBot

Obviously the value of this is huge. Being able to manually coordinate the top point and set the center rotational point r below the top point in post processing would vastly improve the cnc workflow regardless of using luban or fusion360. It would largely eliminate the need to "triangulate" the middle point with the current system (ie, top, left, right).